japaric/stm32f30x-hal #4: Bit-banding GPIO reconfiguration?

idubrov, 2018-01-20:
Why not use ARM bit-banging to allow atomic changes to the GPIO pins configuration without requiring synchronization on MODER/OTYPER?

I can only thing of two drawbacks:

1. you will have to do some memory address calculation math -- but you do bit calculation anyway, to figure out which bits correspond to the pin
2. might be slower in some cases (as each bit requires separate write to certain memory address)

Here is what I did for STM32F1XX

Bit-banding calculation (finding base address in the bit-band region for given peripheral): https://github.com/idubrov/stm32-hal/blob/master/src/bitband.rs

I don't remember, but I think bit-banding is common across different ARM MCUs, so this lower level code could be shared (in embedded-hal or cortex-m crate?)

This is an implementation for STM32F1XX-style GPIO (one that uses CRL/CRH registers):
https://github.com/idubrov/stm32-hal/blob/master/src/gpio.rs

admaunaloa, 2022-01-02:
There is an alternative HAL library that is designed with the advantages of this bitband feature and make it easily usable:
https://github.com/admaunaloa/per_f4_c

idubrov, 2022-01-02:
Since I filed this ticket I've found that bitbanding is now a deprecated feature that is not present in newer cores (presumably, due to higher complexity of caches). Since port writes have dedicated "bitbanding" capability, bitbanding would only help very special cases where port needs to be reconfigured often.

Closing.

closed 2022-01-02 by idubrov