japaric/stm32f30x-hal #6: Help with creating hal for new device

chocol4te, 2018-01-20:
Hello,

I apologise if this is the wrong place for this, but I am trying to build all the necessary crates for an STM32F4DISCOVERY board. I have made a `stm32f411xx` crate using svd2rust and am currently working on a `stm32f411xx-hal` crate. So far I have forked `stm32f30x-hal` but looking through `stm32f411xx`'s lib.rs hasn't made it clear what needs to be changed. Is there any advice you could give me?

I also have a ST NUCLEO-F303K8 that I can build the necessary layers for once I understand the process.

Thanks so much!

japaric, 2018-01-23:
The F3 and F4 are pretty similar so you can pretty much reuse the types; you just have to check that the register manipulation code is correct. I would go at it like this:

- Start with the `stm32f30x-hal` source code
- Replace the uses of `stm32f30x` in it with `stm32f411xx`
- Comment out all the register manipulation code, anything that directly or indirectly uses the `stm32f411xx` crate.
- Start reviewing and uncommenting the commented out code in blocks of functionality. I usually go like this: GPIO, RCC + Timer, Serial, everything else (Spi, I2c, Pwm, etc.).

chocol4te, 2018-01-23:
That’s really helpful, thank you very much!

closed 2018-01-23 by chocol4te