japaric/stm32f30x-hal #22: Getting an additional maintainer for this crate?

byronwasti, 2018-05-04:
It seems like there are a number of outstanding pull-requests with nice features. Would it be possible to get an additional maintainer for this crate so that new features can be added?

TheZoq2, 2018-10-17:
The f103 crate recently got this treatment and I would assume the same could happen here if there are any voulonteers. https://github.com/rust-embedded/wg/issues/224

eldruin, 2019-02-15:
As per [this meeting](https://github.com/rust-embedded/wg/blob/768a21a7d69da766b7191f568df3a7b97bad6d20/minutes/2018-07-30.md) the HAL team is the maintainer of this. Maybe this repo was not setup, though.
Nowadays there is also stm32-rs. Maybe that is also an alternative.
cc @rust-embedded/hal ?

little-arhat, 2019-02-16:
@byronwasti there is also [alt-stm32f30x](https://crates.io/crates/alt-stm32f30x-hal) which does things slightly differently, take a look at the source code (https://github.com/copterust/alt-stm32f30x-hal) or checkout examples [here](https://github.com/copterust/proving-ground).

rubberduck203, 2020-01-04:
Linking related issue #36